hallard/WeMos-Lora #3: Unable to use deep sleep

JaapBraam, 2016-11-01:
I am running a LMIC implementation using this shield. This works fine, but the choice of using GPIO16 for SS breaks the possibility to use the timed deep sleep feature of the ESP8266. In order to make timed deep sleep work you have to connect GPIO16 to the ESP's RST pin. When using this shield, the board won't start up properly when connecting GPIO16 to RST. Maybe, in the next version, the GPIO16 pin can be used for an optional feature so there will be a way to use timed deep sleep.
Especially for battery powered LoraWan nodes a lot of power can be preserved by using deep sleep.

tkerby, 2016-11-24:
I've come across exactly the same issue here. I'd be tempted to wire the button to D0/GPIO16 instead. With a solder bridge option, you can either have deep sleep or a button input. If you select deep sleep then the button becomes an external reset that should cause the deep sleep to end early.

The other option here with the existing boards would be an external watchdog timer IC. The ESP would go into deep sleep then the watchdog would wake it up some time later.  To save pins, the SS line would suffice as a watchdog reset - ie any radio activity keeps the watchdog odd until deep sleep is issued

kragupat, 2016-12-02:
@JaapBraam  @tkerby 
Which LMIC stack are you guys running with this board, I havent had much luck with either of the 2 below
https://github.com/hallard/arduino-lmic
https://github.com/matthijskooijman/arduino-lmic

any help is appreciated.

TheNitek, 2017-04-05:
I just create a fork ( https://github.com/TheNitek/WeMos-Lora ) which removes the LEDs and the button and also frees GPIO16. I just order my board, so I'm not sure yet everything still works, but maybe you want to give it a shot.

hallard, 2017-08-03:
Just released V1.2, if you want to try deep sleep, new PCB revision allow it ;-)

closed 2017-08-03 by hallard