hallard/WeMos-Lora #5: DIO 1/2/3 and Diode 1n4148 - on schematics you have 3 diodes and on assebled picture only one

Netoperz, 2017-02-19:
As in title, the design and assembled dev pictures are different, could you tel me why, and if i want to connect the RFM95 module i should use 3 diodes as i understand, according to the schematics.

SensorsIot, 2017-04-20:
Same question for me. I had to remove the diode to DIO2, because it was always high and the gateway did not work. I also do not understand the role of D1. Is it necessary?
And in my case, the WiFi Neopixel shines red, even if the Gateway works properly.

hallard, 2017-08-03:
Guys
D1 is to lower power of WS2812B to 4.3V like this DI line in drived with 3.3V works, but it's not mandatory, new leds can be drived with 3.3V input when powered at 5V, you just need to try in fact.

To be full LMIC compliant (in term of node) I or'ed DIO0/DIO1/DIO2 with D2/D3/D4 on V1.1 version. May me for GW as said @SensorsIot we need to remove it depending on Server code (management of DIO)

closed 2017-08-03 by hallard