hallard/WeMos-Lora #16: OLED Pins misalignment

hotzenklotz, 2018-07-14:
In v1.3 the holes for the OLED pins are misaligned. The center hole seems flipped and does not align perfectly with the other two PINs. This makes it hard to just "plug and play" a standard breadboard-friendly OLED screen.

Is there any reason for this layout?

[image]

hallard, 2018-09-17:
It's for soldering connector, like that it fit in place when soldering.
BTW you can also just plug OLED onto, it will also stay in place with this format.

closed 2018-09-17 by hallard

hotzenklotz, 2018-09-17:
@hallard Which connector did you use? 

I was hoping to plug [my 4 pin display](https://www.aliexpress.com/item/0-91-Inch-128x32-IIC-I2C-Blue-OLED-LCD-Display-DIY-Module-SSD1306-Driver-IC-DC/32793415973.html?spm=a2g0s.9042311.0.0.27424c4dYRqwrK) directly into the PCB.