hallard/WeMos-Lora #19: Pullup resistors on I2C, why?

atrent, 2019-06-06:
Hi, is there a motive for R1 and R2 (pullup resistors for SDA and SCL)?
I'd like to use Wemos pins D1 and D2 for a SoftwareSerial...
Thanks

pfeerick, 2019-06-29:
Because [I<sup>2</sup>C needs pullups](https://learn.sparkfun.com/tutorials/i2c/i2c-at-the-hardware-level).

If you're not going to use I<sup>2</sup>C, don't put them on.

atrent, 2019-07-10:
thanks for the reply (and sorry for the delay in my follow up)

I actually have an old version of the shield (see photo) and I've checked now that R1, R2 and the 24AA... chip are NOT present, so the fact that I could not activate a SoftwareSerial on those pins is to be investigated more, my bad

[image]