hallard/WeMos-Lora #20: SMD component list

leocopter, 2019-06-10:
Hi, could you provide the SMD component list ( with dimensions) ?
Thanks.

pfeerick, 2019-06-29:
Um... under Bill of Materials on the front readme? 
> all components are 0805 and/or PTH

and 

> Here is the [octopart BOM](https://octopart.com/bom-tool/SJPhS5Am)

leocopter, 2019-06-29:
Thank you.
Do you think this 24AA
02 :
https://m.ebay.ca/orderDetails?itemId=271542332935&txnId=2161023119017
Is the right one ?

Regards,

Chris

pfeerick, 2019-06-29:
Yeah, [that one](https://www.ebay.ca/itm/EEPROM-SERIAL-2KB-SMD-SOT-23-5-Part-24AA02T-I-OT/271542332935) looks like it should be fine... it's the right part number, and it's the five legger S0T-23-5, not the 8 leg SOIC-8 package.