hallard/WeMos-Lora #23: Older Versions

peterl3233, 2019-11-27:
I appear to have a quantity of older "Lora Node Mini" boards for Wemos D1 mini.  I bought them about 2 years ago. They do not have a version number or CH21 logo but have the number WS2812 screened just under the RFM9x moduls pads. The layout and parts placement is identical to V1.3 sans the extra features.
So, Question is:  Is this indeed an early version of WeMos-Lora and is is usable?
Thanks much Peter

TheNitek, 2019-11-27:
Is it maybe the early version of my fork (https://github.com/TheNitek/WeMos-Lora/)? I reduced the original in size an called it mini and reduced it further later on and started to call it micro.
If so it will still work perfectly fine.

TheNitek, 2019-11-27:
Looks like this: https://s3.amazonaws.com/pcbsio/svgs/43aeafab5f987c7c728c912dfaa8ddd3aa04478626dfc0893355db2fd9175ac8/top.svg

peterl3233, 2019-11-28:
Hi Claus; Its slightly different than both examples that you cited. This one has the OLED connector pins.  I found my records and I bought them from PCBS.io. Bought in September 2017. Searching their website no longer finds this one.  Hopefully you can see the attached photo.

[image]

TheNitek, 2019-11-28:
That’s my forked version

peterl3233, 2019-11-28:
Thanks much.  I just bought some RFM95's to play LoRa ... Too much fun!