hallard/WeMos-Lora #25: Ordering new pcbs for WeMos Lora RFM9x Gateway Shield

saysphilippe, 2021-01-12:
Hi, I have tried several times to order your board from PCBs.io - but it looks like they have issues there. Are your boards available somewhere else?

hallard, 2021-01-12:
To be honest I do not have any rewards from PCBs.io since August 2020 so my guess they are not on business anymore because got boards never received

So I added link so you can order the board on [oshpark](https://oshpark.com/shared_projects/E9P1U0Ae)

[image]

saysphilippe, 2021-01-12:
Great, thank you!

henkiejan1, 2021-01-14:
First of all i do not work for TTGO or Lilygo. But i read that TTGO also made pcb's with a minimum quantity of 1. S[ee this from their website.](https://0.rc.xiniu.com/g3/M00/16/67/CgAH513o4-mAIb7uAAHywAhSKd0113.jpg) Maybe a good idea to contact TTGO?